Adhithya03/FOC_DOCS #31: Analyse estimated parameters of IM we did

Anabhayan, 2024-03-30:
https://docs.google.com/spreadsheets/d/1R93BFx19cRxJtGeKPxUc1AqYtrNe2e033AtmHKbUqCM/edit?usp=sharing
Negative rotor resistance!!!

Adhithya03, 2024-03-30:
https://chatkit.app/s/xhe7o6ef3mfwjnj

GPT-4 32k

Adhithya03, 2024-04-08:
[image]

Adhithya03, 2024-04-11:
```
    case 'REVO'
        acim.model      = 'REVO';%    // Manufacturer Model Number
        acim.sn         = '001';      %         // Manufacturer Serial Number
        acim.p          = 2;          %           // Pole Pairs for the motor
        acim.Rs         = 65;         % Stator Resistor
        acim.Rr         = 5.11;       % Rotor Resistor
        acim.Lm         = 1.01;       % Magnetizing inductance
        acim.Lls        = 0.1622;     % Stator leakage inductance
        acim.Llr        = 0.1622;     % Rotor leakage inductance
        acim.J          = 1.75e-4;    % Inertia // measured
        acim.B          = 2.04e-4;    % Friction Co-efficient // measured
        acim.I_rated    = 1.83;       % Rated current (Phase-peak)
        acim.Id0        = 1.08;       % Magnetizing Current (Phase-peak)
        acim.V_rated    = 415;        % Rated voltage (line-line, rms)
        acim.N_rated    = 1440;       % Rated speed (rpm, speed at full load)
        % If data sheet value is not available
        % use ratedslip*N_base
        acim.N_base     = 1500;       % Synchronous speed (rpm)
        acim.F_rated    = 50;         % Rated frequency (Hz)
        ```
        is it ok?